N0BOY/FT8CN #2: Cannot record

MichaelRenner, 2022-12-02:
Moin,

while sending works fine it's not possible to receive. Error message:

Cannot record. Please check permissions. startRecording() called on an uninitialized AudioRecording

Phone is a Samsung S10+. Permission to access the microphone while using the app is granded.

N0BOY, 2023-01-08:
It looks like a problem with App Permission. Can you try to allow microphone again? If it doesn't work, try backup your log and reinstall the latest version.

MichaelRenner, 2023-01-09:
Moin,
thank you for your answer. It is interesting. I installed the new version (0.83). Still no audio (recording). But an error message that the device has permission problems (but everything is allowed).
So I activated my Bluetooth headset. And with that, audio worked. And now it comes: Then I deactivated the headset, audio again via the built-in microphone: And it continues to work!

Now I can keep trying.

There was one crash (Android told me to clear the cache) and a few more crashes, but I don't see how this is related yet.

Micha

PS: I'm going to close that issue.

closed 2023-01-09 by MichaelRenner