N0BOY/FT8CN #9: No source code or license

rogerclarkmelbourne, 2023-01-30:
I can't see any source code in this repo.

Is this an Open Source project, ?

PD5DJ, 2023-01-31:
No open source project.

rogerclarkmelbourne, 2023-01-31:
OK. It seems risky installing an unknown APK with no source code.

How can I scan this for viruses etc

rogerclarkmelbourne, 2023-01-31:
FYI. I scanned the latest release using VirusTotal, and it did not report any viruses. However this only scans for know virus engines, it would not detect things like capturing the shared clipboard and sending that data to the cloud etc

I've not installed it, as I don't have an android VM available at the moment, so I don't know what permissions it requests

https://www.virustotal.com/gui/file/3a9b9bcfdc3f68aedded21eddf662137b2fd15f84ac27d3c600e30c50a8a9a55?nocache=1

PedroPauloBrazil, 2023-02-12:
> FYI. I scanned the latest release using VirusTotal, and it did not report any viruses. However this only scans for know virus engines, it would not detect things like capturing the shared clipboard [...]

Hello, Roger, long time no see!

I have installed this app on my test device and it not only performs very well but appears to be safe too. It requires only 3 permissions: 1) near devces, 2) localization and 3) microphone. 
The absence of a license, English documentation and source code is not so good as desirable, but anyways the app worth a try.

Regards,

Pedro Paulo - PU2OIP

rogerclarkmelbourne, 2023-02-12:
Yes. Being of unknown origin is not ideal.

I will put it onto a small tablet which I do use for email or anything which requires login etc. 

I would not put unknown apps onto any devices you also use for banking etc
(I don't use banking apps myself)

PedroPauloBrazil, 2023-02-12:
Yes, I do not use "hot" devices to test new apps too (I have a segregated account too for this purpose). BTW, this app is very good and I am using it to make my portable operations, with very good results. Congratulations to the author!

And yes, there are some paid wspr and ft8 apps on the store, but I don't have experimented them.
 73!

N0BOY, 2023-02-13:
The developer does have the plan to release source code once the app becomes more mature.

This app is still experimental, please report any bug or security concern you may find, to help with the development.

Btw, let me know if anyone is interested in helping with localizing the documentation.

rogerclarkmelbourne, 2023-02-13:
@N0BOY 

OK. Thanks.

I can't help localisation as I only speak / read / write  English

N0BOY, 2023-02-26:
Source code released under MIT License

rogerclarkmelbourne, 2023-02-26:
wow.

Thanks

closed 2023-03-12 by N0BOY

SpudGunMan, 2023-03-12:
> wow.
> 
> 
> 
> Thanks

Indeed, I didn't get such kindness when I asked. 73