N0BOY/FT8CN #23: Cannot record check permission error

ki5gnl, 2023-03-26:
Love this app. Works great on my galaxy s9 phone but not on my new android 10 tablet or android TV box.  They both give an error of "cannot record. Please check permissions. Start recording () called on an uninitialized audio record"  any thoughts? 
[image]

frohro, 2023-04-16:
I have the same problem with my Pixel 6 Pro.  The Settings says the app has microphone permissions, nearby devices, and location.

frohro, 2023-04-16:
Same thing on an Android 7 phone.  Makes me think I'm doing something wrong.  I just went to the decode with the default settings and tried to decode something.

ki5gnl, 2023-04-16:
I also noticed that it did not ask me which usb device or port like it does on my galaxy so it might not be able  to use the usb serial ports

frohro, 2023-04-18:
WB2CBA said he had the problem, but upgraded the tablet to Android 12 and it worked.  For me, I tried Android 7 and Android 13 with no luck.  I was using the microphone as the input device, or at least that is what I was trying to do.  There is no spectrum display and the mic blinks red, and if I turn the mic off, by touching the icon, and then turn it back on I get the message.

Relayer1974, 2023-05-21:
I had the same issue on a Pixel 4a 5G running Android 13 when only accepting the microphone permission. I had to accept all three permissions upon starting the program - location, microphone, and nearby - to access the mic. I was able to remove the unwanted permissions and it still worked afterward.

bpieres, 2024-02-20:
Same problem! Any update?? Thanks

sq2rby, 2025-07-27:
i have also problem with "Cannot record, please check permissions startRecording() called on an uninitialized AudioRecord" , Tablet Nokia T20 model TA-1397 , Android 13 , kernel 5.4.254 , compilation 00WW_3_590 
Of course all permissions are granted at settings.
Tried off and on permissions without any success.
Please help with this .