N0BOY/FT8CN #24: Ft8 help

Andy71m3, 2023-03-27:
Would love a manual for the app and why isn't anyone coming back to me does the phone speaker have an affect on it using ft817 and Bluetooth
Nothing on psk reporter as well but all seems to decode via ft8 app 

Thanks andy

Andy71m3, 2023-03-28:
I am scratching my head again wondering if....
A radio isn't working 
B propagation is rubbish 
C everyone is using  big power
D Compromised antenna 

Can you spot yourself see if your actually getting out 
Any help greatly appreciated

k9aeg, 2023-07-15:
I'm having  same issues  here. Will do Pskrepoporter.  See if that is a hint.

k9aeg, 2023-07-15:
> Would love a manual for the app and why isn't anyone coming back to me does the phone speaker have an affect on it using ft817 and Bluetooth Nothing on psk reporter as well but all seems to decode via ft8 app
> 
> Thanks andy

Have u had any luck trying to make and ft8 contact yet?

k9aeg, 2023-07-15:
> I am scratching my head again wondering if.... A radio isn't working B propagation is rubbish C everyone is using big power D Compromised antenna
> 
> Can you spot yourself see if your actually getting out Any help greatly appreciated

Did u find problem? Am having  ssme.

chrisdebian, 2025-04-01:
Hi, @Andy71m3 Andy.

Is this still a problem, or can the issue be closed?