N0BOY/FT8CN #31: Yaesu FT-100 profile

Bu2ae, 2023-04-21:
Please release for YAESU FT-100(D) , Because there is no suitable profile for FT-100 rig in the FT8CN. Thanks！

73
BU2AE

Bu2ae, 2023-05-15:
I have try all Yaesu profile(FT-450D, FT-817, FT-857, FT-897, FT-2000...), no one can be used for FT-100D, ha...
Many thanks for your reply！

d3m3vilurr, 2023-11-11:
recently, my local friend got a FT100 for SOTA/POTA activation & he asked to me about FT8CN + FT100.

in my checking, current code set wouldn't support FT-100. [here](https://www.manualslib.com/manual/898069/Yaesu-Ft-100-Micro-Mobile.html?page=99#manual) is CAT command manual of the FT100 and we can find implementation in the [hamlib](https://github.com/Hamlib/Hamlib/blob/master/rigs/yaesu/ft100.c)

(actually we cannot use hamlib code, cuz license of hamlib is LGPL. but, at least, command part is same as manual)