N0BOY/FT8CN #42: (tr)uSDX audio over cat

clewisit, 2023-05-15:
Will it be possible to add support for the (tr)uSDX alpha firmware that supports audio via the cat interface?  This device does not have a built-in usb sound card and I haven't had much success with vox.  Most times the transmit audio goes out of the phone speaker instead of the headphone cable.

M7SVB, 2023-05-17:
I'm fairly certain this would be impossible as trusdx USB interface does not have any connection to the radio's audio input/output.

clewisit, 2023-05-17:
That's not quite true anymore.  

https://dl2man.de/5-trusdx-details/

Unfortunately, I don't understand how the audio stream over cat connection works.

d3m3vilurr, 2023-08-12:
Hello, can someome test this patchset?

[ft8cn-trusdx-debug.apk.zip](https://github.com/N0BOY/FT8CN/files/12327174/ft8cn-trusdx-debug.apk.zip)

patch list:
- split rig (tr)uSDX from Kenwood TS-480
- Initial support voice streaming over CAT for (tr)uSDX
- change http server (for export log) port to 8050 (my phone cannot use 705x port)

requires
- rig volume 9
- CAT + USB mode
- baud rate 115200

problems
1. USB cat signal is pretty unstable. sometimes work well. sometimes disconnected from rig.
2. TX audio freq should be < 2000hz. if over it(>= 2000hz), rig transmit signals to multiple wrong points.
  lower hz also transmit different points, but in these case, these extra signals has very week powers. so it wouldn't be actual problem.

problem 1 could be related my patch. it migth need to wait time between serial communications
problem 2 could rbe related would related (tr)uSDX firmware. idk solution yet.

clewisit, 2023-08-16:
@d3m3vilurr, sorry, I just saw this today.  I loaded up your debug app and connected my (tr)uSDX.  I am able to receive and decode, so this is great!  I haven't tried transmitting just yet, and there are a couple of issues with the test: 
1) crashes frequently, not sure why.
2) adjusting settings (volume, bandwidth, etc.) also seems to cause freezing or crashing in the app.
3) is it possible to automatically set the bandwidth to 3K or 4K?

Everything else seems to work well.  Thanks for your hard work on this.

d3m3vilurr, 2023-08-16:
[ft8cn-trusdx-debug.zip](https://github.com/N0BOY/FT8CN/files/12361159/ft8cn-trusdx-debug.zip)
here is 0.90 mod version
- 
>     1. crashes frequently, not sure why.

sorry IDK.
I didn't get the app crash during change the freq / volume in the app
(I just got a usb disconnecting. and I think it could related cable or device port)

>     2. adjusting settings (volume, bandwidth, etc.) also seems to cause freezing or crashing in the app.

current time, changing volume (signal output strength) of the app is nothing happen to (tr)uSDX.
I didn't apply this multiplier to sending signal streams.

in the trusdx-audio guide, rig volume have to be 10.
I set my rig's volume to 9~10 and mode/bandwidth USB 4K
 
>     3. is it possible to automatically set the bandwidth to 3K or 4K?

probably no.
I cannot find a command about set the bandwidth in Kenwood CAT spec..
I think (tr)uSDX firmware also doesn't support it.

clewisit, 2023-08-16:
Thanks, I will try out the new one later today and let you know my results.

clewisit, 2023-08-16:
Ok, I got a chance to hook this up on my lunch break.  I am able to confirm receive works, it decodes the entire 3Khz spectrum. 
 Transmit also works.  I didn't get any responses, but I confirmed I was heard on PSK Reporter.  

However, the app still crashes frequently.  First time I ran it for about 2 minutes on receive, and then it got through 3 transmit cycles before crashing.  After restarting the app, it immediately crashed.  restarting again and it ran for 1 receive cycle then crashed.  It's fairly random.  I just realized I had turned up the volume so I could see the waterfall better, and I wonder if going above 10 is causing some kind of overflow causing the crash.  I will reduce volume on the (tr)uSDX and try again after work.  If I get more crashes, I will grab the logcat.

d3m3vilurr, 2023-08-17:
yep. please share the logcat if you can grap.
i think, one of possibilities could be related new jni lib.

and thank you for your confirming transmit result.

clewisit, 2023-08-17:
ooh, this was fun.  I decided to run it on my drive home, and was transmitting for about 10 minutes calling CQ.  Still never got a response, but psk reporter has me received all over the east coast US (I'm in California), and got decoded in Mexico, Puerto Rico, Brazil, Hawaii and Australia.  Then it crashed again.  

This time I can confirm it's volume related.  The app only crashes on receive.  When I had the radio on volume 12 it crashed almost immediately, but on 9 it runs about 10 minutes before crashing.  As the volume is decreased, I seem to get less crashes, but also decode less signals.  Sorry, I didn't get a logcat yet, but I will try to get that today.

d3m3vilurr, 2023-08-17:
@clewisit I also retested with my latest debug build during very limited time. (maybe 5min)

here is my app setting
[image]

rig setting was
- volume 15
  - Usually, I set the 9 or 10. it referenced from DL2MAN's video (he set 9) and PE1NNZ's README of the trusdx-audio project. (he suggested 10)
  - this setting is you said bigger volume could occur the crash more easily
- filter 4k0
- other options use default factory setting excepts `8.3 Ref freq`
- power was probably < 2w. I only used USB power from phone.

[image]

results:
I made 2 QSO, but only one is perfectly made. QSO with JF3UFJ was finished with 73. but JG5VFK couldn't.

[image]

don't know the reason, my IC7300 + WSJTX is reported that this signal doesn't have the correct callsign.
it might be related to the time offset or an app couldn't make a correct signal in this case.

TKS DE DS1UFX

clewisit, 2023-08-18:
Alright, the crashing issue is limited to that one phone.  I was able to use my other phone and ran ft8 on 10 meters for about 30 minutes without any crashes.  And, made one contact...while driving to pick up my wife.  It's a good thing this is basically automatic.

Also, the radio volume didn't seem to have any affect on signal this time.  I initially had it at 0 and I could see a bunch of signals.  and put it all the way up to 16 without any change on the waterfall.  So, I think this is good to go.

d3m3vilurr, 2023-08-18:
thank you for sharing the testing result. and congrats made QSO! 👍

N0BOY, 2023-08-18:
Can you guys also test this version? [FT8CN_0.91_Beta1_20230816.apk.zip](https://github.com/N0BOY/FT8CN/files/12381245/FT8CN_0.91_Beta1_20230816.apk.zip)

This build is not a direct merge from https://github.com/N0BOY/FT8CN/pull/67. Instead, @BG7YOZ rewrites the uSDX part in the 0.91 revision. 

Note that, this beta version may require reinstallation. It's better to test on a spare device. And always keep a backup of your QSO log.

d3m3vilurr, 2023-08-20:
@N0BOY it doesn't work.

imo, it would be similar to kenwood 480.

here is test result;
- when the rig was connected, an app seems to send `UA2` correctly.
- The phone mic & speaker are still working. it means an app consider this type still uses `sound` method.
- RX spectrum is wrong. it wouldn't receive from rig or it doesn't resampling the data (RX signal encoded PCM8 + 7825hz)
- TX signal wasn't sent.

d3m3vilurr, 2023-08-21:
@bg7yoz Hello, I used google translate for your word. so I might got wrong understanding.

1. boud rate must be 115200. it was changed.
2. RX audio must be 7825hz  samplerate + PCM8 mono
  and  TX audio must be 11520hz samplerate + PCM8  mono
  it means, we have to do upsampling / downsampling;
  - upsampling for RX: https://github.com/N0BOY/FT8CN/pull/67/commits/ca08aa34 
  - downsampling for TX: https://github.com/N0BOY/FT8CN/pull/67/commits/8bbcc2be
  
  these resampling make the audio quality to be pretty low, but Guido who the man makes a trusdx firmware might think it enough to communicate FT8.
  and it's actually enough to FT8 in my experience
4. `US` message will be finished when you send another command (such as `FA` or `TX`)
  if you don't send any command, stream data never contain `;` (frame data of serial line doesn't have `;`)
  so you are right. my patch will not finish `US` command until receiving result of another command.
  https://github.com/N0BOY/FT8CN/pull/67/commits/98fc73ba#diff-0dcc60d25a9cc310fc7eea760fd06b7163ffbc5b7595a5d2938ddfaf12859a49R119

  so we have simple state machine for current state to detect RX receving

5. basically I also check the `trusdx-audio` project for implementing this and I wrote detail information to first PR description #61

6. i'm not sure. but trusdx wouldn't integrate a good audio compression solution. it needs more cpu power, and trusdx is actually arduino(atmega328p)

d3m3vilurr, 2023-08-21:
@bg7yoz it works well

clewisit, 2023-08-24:
I can also confirm that.  Using 0.91 beta 5, I am able to decode and send.  Finally got New Zealand, with the (tr)uSDX powered only by my phone, and hooked up to the 10m antenna on my jeep.

IU2QMW, 2023-08-30:
> I can also confirm that. Using 0.91 beta 5, I am able to decode and send. Finally got New Zealand, with the (tr)uSDX powered only by my phone, and hooked up to the 10m antenna on my jeep.

Can you post the correct settings for the (tr)uSDX ?
I tried with Xiaomi Redmi note 7, rx sometimes crash, Tex don't work.
Thank you

d3m3vilurr, 2023-09-02:
> Can you post the correct settings for the (tr)uSDX ? I tried with Xiaomi Redmi note 7, rx sometimes crash, Tex don't work. Thank you

my screenshot is correct setting.
but you should use just your callsign without `/QRP`

rmoravcik, 2023-09-03:
Tested beta5 over the weekend. The only problem that I observed was that time to time FT8CN started transmission over phone speakers.

IU2QMW, 2023-09-05:
> my screenshot is correct setting. but you should use just your callsign without `/QRP`

I tried removing /QRP and it works fine.
Thank you

d3m3vilurr, 2023-09-06:
> 

sorry, idk, crash issue too.... it needs crash dump or logcat dumps.
nobody dumped yet.

i just share the correct setting information.
but I'm not re-tested with `/QRP` yet. so I told to him to operate without `/QRP`.

anyway I thought you already tested QRP issue, and imo, full fixing needs to resolve #72

d3m3vilurr, 2023-09-09:
hi BG7YOZ

thank you for updating the package. I'll test at home with beta9 + `/QRP` tag then shere the result in here.

TU DE DS1UFX

d3m3vilurr, 2023-09-09:
1. spectrum data seems too high gain. I'm not sure this is normal or not. a problem is receiving db is reduced than ic7300 datas of wsjt-x.(but it might related rig setting)
    upload images before / after (without antenna)
   
    [image]
    [image]

    here is with antenna
    
    [image]

2. when I set the `/QRP` tag. wsjt-x(v2.6.1) cannot decode the message. (first step; `JA4BXL DS1UFX/QRP PM37` => `<...> DS1UFX/QRP`). so, IMO i cannot met wsjt-x operators
4. QSO is correctly made without any tags. so I can confirm this version is also work well.
5. I found another problem. when I rig set the CW mode, rig cannot change USB mode when I connect to ft8cn then rig will restart few sec later. when I set the LSB or USB manually, problem was gone.

d3m3vilurr, 2023-09-09:
> > hello, @d3m3vilurr
> > 1.FT8CN解码的能力不如WSJT-X。频谱增益的问题是不是与电台有关？

IDK, I just said
1. default spectrum shows high noises (or high RX gain). (please compare first image (resample version) & second image (0.91b9 version)).
   this case is no RX case (not connected antenna)
2. dB value is little defferent between two devices. but it would be normal. the db value can be vary in normal case.

> > 2.没有问题。看您的描述，WSJT-X是对使用/QRP的消息正确解码了，JA4BXL DS1UFX/QRP PM37，其中JA4BXL是使用的HASH，WSJT-X解析此消息，是需要上下文关联才能解析出呼号。也就是说，WSJT-X开机以来，没有收到与JA4BXL呼号有关的消息，所以WSJT-X无法通过HASH推断出呼号。您可以长时间观察，如果在标准消息（i3=1 或i3=2）中出现过JA4BXL后，就可以看到解析完整的消息了。对使用HASH的呼号，使用<>标注。
> > 找到HASH与呼号映射关系的消息： `<JA4BXL> DSUFX/QRP PM37`（注意：在JTDX中也有可能会省略<>）
> > 没有找到HASH与呼号映射关系的消息： `<...> DSUFX/QRP PM37`

removing grid is quite normal.
maybe you would be right. i just monitored on ic7300 + pc without antenna.
(in this time, trusdx is only connected to antenna)

i just curious JA4BXL operator can receive my signal correctly, that because i cannot make qso at this time.

> > 3.略。

> > 4.这个是IC-7300中遇到的问题？

I didn't have experiance with ft8cn + ic7300.
but IMO, it could be related mode changing timing

https://github.com/N0BOY/FT8CN/blob/Dev/ft8cn/app/src/main/java/com/bg7yoz/ft8cn/rigs/TrUSDXRig.java#L308-L309
IMO, it should be

```
                    getConnector().sendData(KenwoodTK90RigConstant.setTS590VFOMode());
                    this.setUsbModeToRig();
                    getConnector().sendData(KenwoodTK90RigConstant.setTrUSDXStreaming(true));
```

> > 关于0.91 Beta 9，现在迫切需要了解：(tr)uSDX使用 audio over cat 能否正常工作，因为我重新写了Resample的库，替换掉jackz314的aar库。

yeah. it works well.
`/QRP` tag could be test yourself. it just need to reduce tx power to 5w. so i can believe you that this is no issue :)

only remain just 2 problems
1. spectrum; it's just needed to confirm by you that it's normal or not.
2. reboot a rig(it means, a rig was crashed) with CW mode after connecting the ft8cn

> > thanks！
> > bg7yoz
> 
> 补充一下说明，非标准呼号CQ另一个呼号时（非标准消息 i3=4），是没有网格的。 `<JA4BXL> DSUFX/QRP PM37`的实际信息是： `<JA4BXL> DSUFX/QRP`

TU DE DS1UFX

d3m3vilurr, 2023-09-10:
> hello, @d3m3vilurr
> 
HI BG7YOZ

> 0.91b9可以在(tr)uSDX audio over cat正常工作，对我来说是个好消息，意味着这部分的工作完成了。
> 
> 还需要进一步沟通的问题：
> 
> 1.关于频谱高增益。还需要确认： a.第一个图片和第二个图片，是使用ic-7300电台，还是(tr)uSDX电台？ b. 你说的“resample version”是指0.91b5?
> 

both are (tr)uSDX. it was my build but probably same as 0.91b5
BTW i'll retest with 0.91b5 to confirm.(I remember that was same as old one... anyway)

> 2.关于电台重新启动的问题。应该是(tr)uSDX的固件问题。我按照你的提示，在0.91b10中增加模式设置（指令是`MD2;`）
> 

yep. this is what i said :)
I retest it to check the fixing rig crash.

> 
> 73！
> 
> bg7yoz

73 TU
DS1UFX

d3m3vilurr, 2023-09-11:
> HI BG7YOZ
>
>
> both are (tr)uSDX. it was my build but probably same as 0.91b5 BTW i'll retest with 0.91b5 to confirm.(I remember that was same as old one... anyway)

here is beta5 + (tr)uSDX
[image]

here is beta10 + (tr)uSDX
[image]

both are same RX condition. (just connected USB cable & dummy load)

> 
> > 2.关于电台重新启动的问题。应该是(tr)uSDX的固件问题。我按照你的提示，在0.91b10中增加模式设置（指令是`MD2;`）
> 
> yep. this is what i said :) I retest it to check the fixing rig crash.
> 
an app can change to USB mode when connect the rig. but it still occur rig crash.
we might need some extra waiting time to receive rig answer or it would be just trusdx firmware problem.
IDK actual reason, I'll retest own myself after release the latest codeset.
IMO, tt wouldn't be critical issue if we share the temporary solution to users.

current temporary solution is;
- set rig mode to USB mode + 4k bandwidth first
- then connect to ft8cn

> > 73！
> > bg7yoz
> 
> 73 TU DS1UFX

d3m3vilurr, 2023-09-11:
> 看来封装的重采样效果不是很理想 :-( ，暂时就这样了。
> 
> 这是我封装sample的代码： [resample_JNI.zip](https://github.com/N0BOY/FT8CN/files/12575838/resample_JNI.zip)
> 
> resample代码的来源：https://github.com/intervigilium/libresample
> 
> 0.91马上就会发布，您可以在新的版本修改了。
> 
> 73!
> 
> bg7yoz

can i know function chain of TX/RX.
trusdx only use mono PCM8 for CAT audio signal. so transform PCM16<->PCM8 could make a problem.
but IDK detail reason yet..

N0BOY, 2023-09-12:
@d3m3vilurr Updated v0.91 src in both branches. Feel free to submit PR. Thanks!

clewisit, 2023-09-13:
> 关于crashes的问题，一般是出现在数据处理出错，在java层通过logcat比较容易查出来，我观察了jackz31的resample的库，他是使用了jni，如果是在.so中出现错误，错误很难查出来，这也是我重写resample库的原因。
> 
> 不同的手机，安卓的版本不同，对.so中的内存引用检查的严格程度也不同，有的安卓版本数组越界不会crash，而有的安卓版本出现数组越界就会crash，一般来说，安卓版本越高，对内存管理越严格。
> 
> 这可能是就是有的手机会crash，而有的手机不会crash的主要原因（安卓版本不同）。大概率是.so库运行时出现的问题。
> 
> bg7yoz

Thank you for the insight.  I tested beta 10 on the phone that previously suffered from random crashes with beta 5, and I can confirm it no longer crashes.  I left it running for about 1 hour at max volume.  

I also see the increased background noise in the waterfall.  For some reason this phone is sensitive to volume changes on the (tr)uSDX, so when I turn the volume all the way down it shows a good picture of the rhythm of this noise.  

[image]

You can see the horizonal and vertical bands clearly in the waterfall.  Let me know if you need any further testing.

d3m3vilurr, 2023-09-13:
> @d3m3vilurr Updated v0.91 src in both branches. Feel free to submit PR. Thanks!

yes. I see it yesterday. I cannot test until friday.
I'll recheck the remain issue.

> I also see the increased background noise in the waterfall. For some reason this phone is sensitive to volume changes on the (tr)uSDX, so when I turn the volume all the way down it shows a good [...]

oh... an app might forget turning off the phone voice receiving after the connection. thanks. i'll also test & check it.

d3m3vilurr, 2023-09-16:
1. if rig uses cw mode, a rig is always crashed after connect to ft8cn. i'm not sure actual reason, but it could be (tr)uSDX firmware issue.
2. rx stream data + waterfall is not related phone volume. but related rigs volume. users have to reduce the rig volume to community's suggest level; 9 or 10.
3. @bg7yoz can you wrap [libsamplerate](https://github.com/libsndfile/libsamplerate/tree/master/src)? noise of the waterfall can be highly related `resample` logic. [stanford's resample](https://ccrma.stanford.edu/~jos/resample/Free_Resampling_Software.html) is too old. i'm also not expert for the resampling logic. but imo, this lib's old bug which already resolved in the libsamplerate or unsupport things such as channel make a problem.

rmoravcik, 2023-10-01:
> hello, @d3m3vilurr
> 
> 我按照你的要求，重新封装了libsamplerate，请测试能否正常工作。
> 
> [FT8CN_0.91p1_New_Resample.apk.zip](https://github.com/N0BOY/FT8CN/files/12647687/FT8CN_0.91p1_New_Resample.apk.zip)
> 
> 73！
> 
> bg7yoz

Hello bg7yoz,

I was testing this new build over the weekend and I didn't observed any crash. Waterfall doesn't seems to be overloaded with antenna attached and volume set to 10. 

I only observing disconnection from rig. There is an popup saying something like: "Connection with rig is broken. ... 2 > 1". Sorry I don't have an screenshot of it.

Thank you,
Roman

k6tw, 2023-10-05:
OK, I just found this conversation and I am having similar problems using trusdx with FT8CN. My original conversation is here https://github.com/N0BOY/FT8CN/issues/80 

So, I've had no luck with .91, patch 1 release, beta 10, beta 9, and beta 5- here are my results:

When setting trusdx as the radio- I get the connection error as soon as I select the serial port (I am able to decode only before selecting the serial port).
When setting Kenwood TS480 as the radio- I can establish a connection and decode. I am able to PTT momentarily, but no output power (even though the OLED on the trusdx shows "T") and the rig would just "click" for one second, then the TX audio was through the phone only. Unfortunately, once this happened, I also lost RX audio in the rig, even though it was working before. 

I've spent hours trying to get my trusdx to transmit FT8. Any other suggestions?

Tim
K6TW

d3m3vilurr, 2023-10-05:
@k6tw if you select TS480, you might need USB audio sound card

can you recheck your firmware version? this feature requires latest [experimental firmware](https://dl2man.de/wp-content/uploads/2022/01/wp.php/beta.html) not stable firmware.

k6tw, 2023-10-05:
My firmware is 2.00u

Thank you,

Tim

starcrafti, 2023-10-11:
我的usdx添加了蓝牙模块。在.90下，由于电台型号中没有usdx，选择了QRP Labs QDX，cat-蓝牙连接下可以自动发射，电流增大、屏幕显示T，持续数秒。而在.91 patch1下，电台型号选择trUSDX，不管控制方式是选择vox，还是cat-蓝牙连接，都无法自动发射，电流没有变化、屏幕也没有显示T。不知是不是什么设置不正确？

d3m3vilurr, 2023-10-11:
@starcrafti IDK, USDX case & I don't have it. AFAIK, their firmwares / rigs are not [official usdx firmware](https://github.com/threeme3/usdx).
and also PE1NNZ isn't release trusdx's TX part on the firmware code.
(imo, developers can implement it, but idk current state)

so, you can't use `(tr)uSDX` setting, you should use `Kenwood TS-480`. and also, bluetooth CAT line wouldn't broadcast TX/RX stream. if you still cat the empty voice when connect bluetooth CAT + TS-480 mode, you have to need external device such as USB audio stick.

starcrafti, 2023-10-12:
@d3m3vilurr Thank you! When I chose TS-480, it did work, as if at 0.90. At present, due to the poor electromagnetic environment in my location, I am not sure if the automatic launch is working properly（*/∇＼*）Thank again!

k6tw, 2023-10-13:
So has anyone successfully been able to use audio over CAT (only the USB cable) using a trusdx and FT8CN? I've spent many, many hours troubleshooting and still can't get it to work (I mentioned earlier, the no transmit problem). If I can't get an answer on the support forum here, I don't know what else to do. I hate to buy a digirig, but it looks like I may have to do that.

starcrafti, 2023-10-13:
> 您好 @starcrafti ：
> 
> 正如@d3m3vilurr所说，在FT8CN的0.91中，(tr)uSDX的电台选项，是采用audio over cat的，也就是音频信号通过串口传输的，您使用的USDX蓝牙方式，建议还是电台还是选择TS-480。 在下一版本中，(tr)uSDX将有两个选项做区分：(tr)uSDX audio over cat和(tr)uSDX(TS-480)。
> 
> bg7yoz

膜拜大佬^_^
目前我选择480，可以发射——用了一个手台接收到信号。不幸的是，从手台里传出的声音，没有ft8那独特的音频声。
也许这可以解释为什么我的呼叫都没有被应答😂
期待新版本。

rmoravcik, 2023-10-13:
> So has anyone successfully been able to use audio over CAT (only the USB cable) using a trusdx and FT8CN? I've spent many, many hours troubleshooting and still can't get it to work (I mentioned [...]

For me it's working except occasional disconnection error. Two weeks ago I did ~100 QSOs during weekend using just USB cable.

k6tw, 2023-10-13:
> 膜拜大佬^_^
> 目前我选择480，可以发射——用了一个手台接收到信号。不幸的是，从手台里传出的声音，没有ft8那独特的音频声。
> 也许这可以解释为什么我的呼叫都没有被应答😂
> 期待新版本

When will the new version be released?

d3m3vilurr, 2023-10-14:
@k6tw i'm a patch author and i tested qso without audio cable.
sorry i also dont know a reason. it can be occured from multiple reasons such as my original codeset, firmware, specific rig type(only reported works with classic & low band), UTG adapter, USB cable...

can you test on the pc with [trusdx-audio](https://github.com/threeme3/trusdx-audio)?

imo, next new ft8cn version wouldn't be different that just split rig type to prevent distinguish.

k6tw, 2023-10-14:
> @k6tw i'm a patch author and i tested qso without audio cable. sorry i also dont know a reason. it can be occured from multiple reasons such as my original codeset, firmware, specific rig type(only reported classic & low band are work. you are first person who report high band rig is not working), UTG adapter, USB cable...
> 
> can you test on the pc with [trusdx-audio](https://github.com/threeme3/trusdx-audio)?
> 
> imo, next new ft8cn version wouldn't be different that just split rig type to prevent distinguish.

Maybe that's the problem. My radio is the high band trusdx (10, 12, 15, 17, 20). I will try connecting to wsjtx next.

rmoravcik, 2023-10-17:
> > > So has anyone successfully been able to use audio over CAT (only the USB cable) using a trusdx and FT8CN? I've spent many, many hours troubleshooting and still can't get it to work (I mentioned earlier, the no transmit problem). If I can't get an answer on the support forum here, I don't know what else to do. I hate to buy a digirig, but it looks like I may have to do that.
> > 
> > 
> > For me it's working except occasional disconnection error. Two weeks ago I did ~100 QSOs during weekend using just USB cable.
> 
> 引起断开的原因，很有可能是天线不平衡造成的射频干扰。

I did an test in "practice mode" enabled. This means, that transmitter is disabled during transmission and not emitting any radio signals and disconnection happened also in practice mode. So, the reason is probably something else.

If you have some other idea or you can add some more traces, I'm happy to test it and reproduce. Thank you.