N0BOY/FT8CN #49: Q900 v4 CAT will not work

blueskiesoky, 2023-07-10:
It decodes audio. It will not transmit through radio. Phone just sounds through phone speaker. Doesn't seem to change frequencies from app either

HB9HXH, 2023-07-10:
I have the same issue with XIEGU X6100. I have tried several tablets and phones, without success. With my REDMI Tablet though, I have added the Bluetooth connection and selected Bluetooth and (after pumping up the volume to max) I've been able to transmit. Note that for some reason the XIEGU X6100 crashes sometimes, but other than that it works damn well: it made two QSOs by itself while I was still checking if it was working :-)

HB9HXH, 2023-07-16:
> 连接android设备应当使用OTG转换头。 Q900建议您使用蓝牙方式连接FT8CN。

Thanks a lot for your kind reply! I'm trying to connect with a Xiegu X6100 with latest FW.
Yes, I tried with two different OTG adapters and I can select the serial links proposed, but in any case the audio for transmission doesn't go through (but I can receive very well and change bands from the App)
When I transmit the power remains 0W and there's no modulation at all with the USB cable.
If I connect bluetooth (in addition from the USB/OTG cable) and set the control to bluetooth I can sometime transmit with modulation (if the sequence of activation is the right one).
If I just connect Bluetooth, without any USB cable, the rig is controlled, but the audio decoded is ambient noise caught from a microphone (the tablet, I suppose) and there's no modulation when transmitting.

blueskiesoky, 2023-08-13:
Thanks bg7yoz.
Will try that and let you know if it works for me.