N0BOY/FT8CN #54: Three problems.....

ZB2ham, 2023-07-15:
A few issues......

**First issue.** The QSO log records errors in the names of countries worked. See the attached files, where many of the prefixes recorded do not correspond with the countries identified. There were many more such occurrences, but this is sufficient to illustrate. For example, in the first image...

N9 is identified as European Russia when it should read USA
KB6 is identified as Asiatic Russia when it should read USA                       
JK3 is identified as Poland when it should read Japan etc etc

The country is not recorded in the adi file, so this should not be an issue with electronic logbooks - I think.

**Second issue.** IM76 is my GRID square, not my GIRD square.

**Third issue.**  I have difficulty accessing my log and other files in my PC browser, as instructed in FT8CN. MOST of the time I simply cannot make the connection. What can I do to make a reliable connection? Mostly "192.168.1.5 has taken too long to respond". I have looked for but cannot find the files in my smartphone. I have made hidden files visible, but still I cannot locate them. If I could locate them I could transfer them easily.

[image]
[image]
[image]
[image]

Rafltdcpa, 2023-07-15:
Concerning "Third issue".....did you add the port number 7050 to the device ip address (ie. 192.168.1.5:7050)?

ZB2ham, 2023-07-16:
I have tried 192.168.1.5 and 192.168.1.5:7050, both many times, and both equally difficult to reach. In all cases my laptop and smartphone have been connected to my only home network. In both cases the message is the same;

"This site can’t be reached192.168.1.5 took too long to respond"

As I said, my problem could disappear if I knew where in my 'phone the files are stored, then I could transfer the adi file manually - but I cannot locate them.

Many thanks,

Ken

ZB2ham, 2023-07-16:
_The first question: the database of the location attribution in FT8CN is correct, because the Android RecyclerView control uses a memory multiplexing mechanism, and FT8CN does not use reasonable codes, resulting in the reuse of other Items without updating the location information in time, and the actual display It becomes the location information of other QSOs. This problem will be corrected in the next version.
The second problem: typo, Gird--->Grid. Corrected in the next version.
The third question: I have not encountered such problems during development. Before using FT8CN, please close all background apps. Maybe other apps occupy port 7050? Please confirm whether the IP address of the mobile phone is correct, and try to write the entire access address: http://xxx.xxx.xxx.xxx:7050. FT8CN does not save the QSO data in the mobile phone as an adi file, but saves it as a database (sqlite format). The file name is data.db. You need to use a database access tool to view the data._

Many thanks, v helpful!

The mobile phone I use is a 'phone which I used previously (Samsung J5), a year or two ago and is running Android 6. I did a factory reset before installing FT8CN, and this is the only purpose I put this phone to.

I have just quizzed ipconfig with the command "arp -a" and I have nothing on 192.168.1.5, and nothing using port 7050.

Ken
[image]

ZB2ham, 2023-07-16:
_Maybe a network problem?
You can use the browser of your mobile phone to visit http://127.0.0.1:7050 (do not close FT8CN) to see if you can access FT8CN. If it can be accessed, it is a network problem. If it cannot be accessed, it is very likely that FT8CN has not started the HTTP service (if this is the problem, there is no solution for the time being :-( )._

Well, the 'phone can see itself! Which is an end to the problem, because I can now download the file to the 'phone itself and from there I can put it anywhere else! Excellent! problems solved - except I appear to have a network issue to resolve....

Many thanks indeed,

Ken

chrisdebian, 2025-04-01:
@ZB2ham Hi, Ken.

You raised three issues in one report. Please could you check to see which if the problems are extant, and raise them as separate issues? This helps meaningful and coherent discussion and hopefully resolution.

Thanks,

Chris

ZB2ham, 2025-04-01:
The only issue remaining is the one I raised last, most recently as No 138
on 26th February.

Many thanks.

Ken Ruiz
ZB2MD