N0BOY/FT8CN #60: Ic-7300

RU0AJQ, 2023-08-04:
Кто-то уже писал про ic-7300, аппарат прекрасно управляется по usb, но периодически он не может передавать, хотя динамик в программе показывает наличие передачи, в чем проблема? Пробовал на планшете и телефоне, производители устройств разные

Rafltdcpa, 2023-08-09:
I have a different issue still unresolved but the symptom seems the same....the issue presents after one or two transmission cycles. It was suggested that common mode RFI was the cause. You might try ferrites on the antenna feed and all cables to see if that solves it. It didn't work for me.

felangga, 2023-10-23:
> I have a different issue still unresolved but the symptom seems the same....the issue presents after one or two transmission cycles. It was suggested that common mode RFI was the cause. You might try ferrites on the antenna feed and all cables to see if that solves it. It didn't work for me.

Already have the ferrites on the cable, but still the same issue. 
Maybe related to the transmitted command to turn on the PTT on the Icom 7300. 
I still trying to debug it though.

felangga, 2023-10-23:
> Пришлось перейти на ft-857 с соединением по аудио джеку 3.5 мм на выездах на природу, похоже автор не может решить эту проблему

Recently (around 10 minutes ago) I might solve this issue by adding a `Thread.sleep(500)` before putting the PTT on. 
So far it's been almost a minute running okay.

felangga, 2023-10-23:
@RU0AJQ @Rafltdcpa I just build my fix, if you want to try if it works on your side or not
https://drive.google.com/file/d/1r5JgN4TaoXC3DT8vTL75QjeR2VuqnVcM/view?usp=share_link