N0BOY/FT8CN #68: Audio output is attenuated after first transmission

gnotu, 2023-08-15:
Some of us on the QRPLabs forum have noticed that a problem recently appeared when using FT8CN 0.89 with QRPLabs QMX transceivers.
(See https://groups.io/g/QRPLabs/topic/100738548)
At least two of us are running FT8CN on Google Pixel 6A Android phones.
We see that the first FT8 transmission happens normally.
But after that every subsequent transmission has no or little audio.
This continues until exiting FT8CN.
After re-running FT8CN, the first transmission is normal again and every transmission after that has low/no audio.
This seems to have started happening pretty recently so it may be due to an Android update.
I am willing to help troubleshoot it on my phone if you are unable to reproduce it.
Thank you and 73.

--Mike

N0BOY, 2023-08-16:
Mike, was everything running smoothly on v0.89?

Let's try the following to troubleshoot the problem. 

1. Update to v0.90 and try with same Android device and radio

Remember to backup your log file before updating, in case you need to rollback. 

2. If problem persist, run FT8CN using the same phone/tablet but with different radios, or same radio with different Android device.

gnotu, 2023-08-16:
@N0BOY everything was fine on 0.88.
Also, 0.89 was fine until recently.

@bg7yoz I will try your suggestions and report back.

73, MIke KK7ER

gnotu, 2023-08-16:
I tested FT8CN 0.90 on a Google Pixel 6A and a QDX transceiver and it works fine.
I tested FT8CN 0.90 on a Google Pixel 6a and a QMX transceiver and it exhibits the problem.
I tested FT8CN 0.90 on a Samsung Galaxy A03s and a QDX transceiver and it works fine.
I tested FT8CN 0.90 on a Samsung Galaxy A03s and a QMX transceiver and it exhibits the problem.

So...  it looks like a QMX issue.

73, Mike KK7ER

gnotu, 2023-08-29:
The designer of the QMX is going to investigate.
https://groups.io/g/QRPLabs/message/108670

73, Mike KK7ER

gnotu, 2024-03-10:
With FT8CN 0.92 and QMX firmware 1_00_017 this issue has been resolved per my testing.

73, Mike KK7ER

closed 2024-03-10 by gnotu

sq5jup, 2024-07-07:
Had the same issue when I try "OOtB", FT8CN 0.92 + latest QMX firmware.
It start to work on my Mi 9 when switched "Adjust media sound in multiple apps" at Sound assistant configuration.
Hope you can do the same on your mobile.

73 de SQ5J,
Piotr