N0BOY/FT8CN #71: Three more issues...

ZB2ham, 2023-08-22:
Dear BG7YOZ

I have been using FT8CN for some months now. I have had some problems, and these are now resolved. FT8CN is very impressive. I am amazed at what it does, and so well. What follows is not criticism, merely suggestions for a better (for me at least!) user experience.

There are three items;

1.	The addition of a **free text option**. I appreciate FT8 is not a conversation mode, but I often have a pile-up and it would be useful to send a message such as “ZB2MD QRT” or “ZB2MD  QSY 21.300” which of course also implies a change of mode to SSB.

2.	**Non-scrolling**. Occasionally the FT8CN does NOT scroll, and it might take a few cycles to notice, by which time the QSO has moved on. I might be sending RR73 repeatedly despite the other station having sent 73s.

The most serious of the issues reported here;

3.	**FT8CN takes control!** When the bands are quiet with few stations, FT8CN does an excellent job automatically, tracking the progress of a QSO and sending the next appropriate message. When I have a pile-up, it is a completely different matter.

I may be in QSO with Station A, and I am ready to send “Station A ZB2MD R-10”. Just before that message gets transmitted, Station B calls me and FT8CN IMMEDIATELY changes from the QSO with Station A and begins a QSO with Station B. 
I then find myself holding two QSOs at the same time. The only way I can stop this is to stop transmission so that the string to Station B is NOT transmitted, but I have not transmitted to Station A either. 
I prepare to send the wanted string to Station A, only to find that FT8CN at the last moment chooses to respond to Station C who has just called me.
At times when I have chosen NOT to correct this, to see what happens, I find FT8CN has started numerous QSOs but does not progress with any of them, as long as new stations keep calling.

Is there a means of locking FT8CN to the QSO that I choose to continue with, rather than have FT8CN override my choice?

Many thanks for your patience and attention.

ZB2ham, 2023-08-25:
As ever, your responses are prompt and comprehensive.

1. I see that free text is available, many thanks - I had not discovered it
2. The non-scrolling may be down to me - I shall analyse what I do more closely to see exactly under what conditions it happens.
3. I look forward to a better implementation of FT8CN taking control, in later updates.

All the three issues I had raised in an earlier post have been sorted with the release of v0.90

Many thanks again,
Ken Ruiz,  ZB2MD

chrisdebian, 2025-04-01:
@ZB2ham Hi, Ken. Does this issue need to stay open?

ZB2ham, 2025-04-01:
No indeed not.

Many thanks

Ken Ruiz
ZB2MD