N0BOY/FT8CN #76: v.0.91  No TX audio with  X6100 firmware 1.1.7.2 (September 2023)

djbr1, 2023-09-13:
Hello
With FT8CN v.091 and x6100 firmware 1.1.7_2 (sept 2023) CAT control and RX works.
Unfortunately, TX audio does not work. Reply in #69 says it is x6100 audio problem 
I managed to have WSJT transmit audio through WFview on laptop using WFserver on x6100 by tweaking alsamixer (Change from DAC to Mixer and increase Volume of Mic1 Boost)
```amixer -c 0 set 'Headphone Source' 'Mixer'```
```amixer -c 0 set 'Mic1 Boost' 24db ```
This can also be done from alsamixer
This tweak allows WSJT or JTDX to transmit through WFview and WFserver

Problem with ft8cn v.091 is that TX power is completely lost after connecting and testing ft8cn,  only factory reset of x6100 brings back tx power.

Rafltdcpa, 2023-09-15:
I'm having the same problem, no transmit audio, using X6100 fw v1.1.6 with FT8CN  v.90. Going to try a factory reset. If that fails I'll reluctantly reflash with v1.1.7, fingers crossed.

Rafltdcpa, 2023-09-16:
Update:

Remaining on FT8CN v.90 I performed a factory reset which resulted in no change.
I flashed the radio with the new FW 1.1.7 (8/31/23), performed a factory reset as suggested here, and that solved the issue.
I think I'll stay with v.90 for the time being. If it ain't broke no more, don't fix it.

Rafltdcpa, 2023-09-17:
I read the v.91 release notes and will probably update.
Thanks for commenting and for this excellent app.

Rafltdcpa, 2023-10-13:
FYI, I upgraded to v .91 a few weeks ago and the CAT and transmit audio seem to be working fine with the R1CBU alternate firmware. I believe that was also the case with the stock Xiegu firmware.

djbr1, 2023-10-13:
> FYI, I upgraded to v .91 a few weeks ago and the CAT and transmit audio seem to be working fine with the R1CBU alternate firmware. I believe that was also the case with the stock Xiegu firmware.

Hello,
Using FT8CN with usb cable has no TX issues. My problem is related to WLAN connection from Android to X6100, where FT8CN connects wirelessly to WFserver on X6100.
I am not using USB cable as X6100 is draining power from Android phone battery, this is a flaw (or feature) of USB OTG connectivity
Thats why I desperately wait for FT8CN and X6100 to communicate wirelessly

Rafltdcpa, 2023-10-13:
Congrats on getting the WLAN facility working at all. I've been
unsuccessful in that regard so far.

Rafltdcpa, 2023-10-13:
With regard to the draining of the phone battery see the pic attached. I'm using a Samsung Note 8 which has a usb C port. The splitter is a usb C male to usb A female/usb A female. This allows for CAT control and the simulcharging of the phone.

[image]

Rafltdcpa, 2023-10-13:
Typo correction....the splitter is usb C male to usb A female/ usb C female ( not usb A).

djbr1, 2023-10-13:
Hello,
I also have split OTG cable. It provides power to phone, however it does not prevent X6100 from draining from phone :(
BTW, it is X6100 behaviour with all attached devices (laptop, PC), approx 100-200 mA are drained from attached USB.
If you have current meter on your power supply, you can notice the difference in consumption when USB is attached or not
73 HB9TXB
NB: you can join discord server where x6100 is dissected past two years ( https://discord.gg/VD4AsYJH  )

Rafltdcpa, 2023-10-13:
The power supply I use is just a simple 12v/10amp wall brick with no indicators so if there is a current drain I haven't noticed it. The phone always reflects a battery charge of 100%. 
I would be curious to know if, when you used the X6100 with the usb cable and experienced the phone current drain, if the radio was using external power or battery power and, if battery power, was the charger set to on or off.

djbr1, 2023-10-14:
Hi @Rafltdcpa 

I made measurement from my 12V/30A power supply, ampermeter precision is 0.1

	                  laptop connected USB	laptop disconnected USB
radio settings charge on	0.3 A	               0.5 A
radio settings charge off	0.2 A	               0.4 A

Drain of 0.2 A from USB is visible in all scenarios

As I do not have ampermeter for USB cables, it was not possible to measure drain when X6100 runs on battery

73

djbr1, 2023-12-28:
Hi @bg7yoz 
If you have contact with Xiegu developers, please ask when new FW will be released.
thanks
 
> 请耐心，XIEGU的新固件一定会支持FT8CN+WLAN的。期待x6100新固件的发布。

dk5cr, 2024-10-01:
hi, 
current Firmware for Xiegu x6100 is released. Firmware version is 1.1.8., baseware is 1.1.7. 
Unfortunately audio issue isn't fixed as far as i could try. 
Cheers 
Cornel

oyidi, 2025-02-04:
> hi, current Firmware for Xiegu x6100 is released. Firmware version is 1.1.8., baseware is 1.1.7. Unfortunately audio issue isn't fixed as far as i could try. Cheers Cornel

You can try this unofficial firmware: https://github.com/gdyuldin/x6100_gui
It works good for me with JTDX.

dk5cr, 2025-02-04:
Thx Oyidi for you post. JTDX is an APP on the cellphone? Does it work with FT8CN via bluetooth also? 

Just downloaded the current version, but on [app 3:3] there's only "WiFi" visible to switch. When i activate this button, then WiFi/BT: is set to "On". But there is no chance to connect a BT-device even there's no list of accessible BT-devices. 
Also on the cellphone the x6100 doesn't appear as BT-device :-(