N0BOY/FT8CN #80: Serial Port error/no transmit with trusdx

k6tw, 2023-09-25:
Hello, when I connect to the trusdx, I am able to see the decoded signals. I am unable to transmit and PTT does not work. When I choose the serial port, it connects successfully, but within one second, I then receive a message saying, "Communication error with rig, the connection is broken".  Then some Chinese writing and at the end of that is "2>1"

Using:
trusdx firmware 2.00u (latest) 
Samsung S23 Ultra (updated)
Baud rate/connection- 115200, CAT
50 ohm coax to steppir antenna (very good swr)
USB cable: https://www.officedepot.com/a/products/795902/Ativa-USB-20-Type-C-to/?utm_source=google&utm_medium=cpc&mediacampaignid=71700000113868766_20488703163&utm_source=google&utm_medium=cpc&gclid=Cj0KCQjwvL-oBhCxARIsAHkOiu19jyyzvgNkTaNjuiCrhvcJXwZQzdJV2X4VasF4A2ZKWersmbSuwJwaAu00EALw_wcB&gclsrc=aw.ds 
[image]

d3m3vilurr, 2023-09-25:
can you share this?

1. testing result with rig sound is 9~10
2. testing result with sample rate is 12kHz

btw, idk actual reason, when I operate on the higher freq (15m~10m band) with x6100,
i also got a broken connection.
(at that time, only 30m band was bad swr, other bands are good)

it might be related RFI.
One of korean operator uses ferrite choke to usb cable at the POTA activation, he shared result that it could resolve(or reduce) the problem.

k6tw, 2023-09-25:
Thank you for the suggestions. 

1)- Rig with sound at 9 or 10- does not solve the problem
2) sample rate at 12khz- does not solve the problem

The rig gives me the same error at 20 and 10m, so it is not a band specific problem.

It is not RFI related because I am not able to PTT and there is never any transmit power, so therefore no RFI. 

Thank you,

Tim
K6TW

d3m3vilurr, 2023-09-25:
did you only get a problem on the 20m and 10m? or all bands has same problem?
is your rig classic band version?

k6tw, 2023-09-25:
I have the problem on all bands. My rig is the high band version- 10, 12, 15, 17, 20meters.

d3m3vilurr, 2023-09-26:
i re-read the issue. sorry i was read your comment very roughly.

can you share your default mode of the rig?
IDK reason yet, when an app connect to the rig with non-USB mode(such as CW), rig will reboot then restart with old mode.

so if you set cw or other mode, please set the USB + 3K0 filter

immediately disconnection could be related this issue, phone, UTG adapter and cable.

k6tw, 2023-09-26:
I can confirm the mode is Usb with bandwidth of 3k (or 4k, i forgot which one).

Thank you,

Tim

k6tw, 2023-09-28:
Just tried a new cable. Unfortunately, same problem. Is there anywhere else I could post in order to get some help on this?

clewisit, 2023-10-04:
It just occurred to me, I've never asked which version of ft8cn you were using.  I also had connection issues with earlier betas, but the beta 5 version worked on one of my phones while it would crash on another phone.  Try using some of the earlier betas.

k6tw, 2023-10-04:
Thank you for the info. I'm using release. 91, patch 1, which should be the newest one.

Can you clarify about the different betas? I only see .91, .90, .89, etc.

Do you mean just try the previous release,  like try .90 instead of .91?

Thank you,

Tim

clewisit, 2023-10-04:
No, I mean go to the issues tab.  Look for issue #42   
Read the discussion.  There will be some test files there.  Try them out and see if you have different results with any of them.