N0BOY/FT8CN #113: Issue using Bluetooth devices

iz0zpb, 2024-07-14:
Hi all, I have a problem when using a Bluetooth device for audio in TX and RX. The Bluetooth device I am using is a TSA6015 (the one in the photo). If I configure this device to handle both calls and multimedia audio (through Android settings), FT8CN decodes the audio, but when it goes into TX, it does not send the audio to the Bluetooth device but to the smartphone (this should be the correct Bluetooth configuration); if I configure the TSA6015 (still through Android settings) to handle only multimedia audio, it works well in TX, but not in RX as FT8CN takes the audio from the phone and not from Bluetooth; if I configure the TSA6015 (again with Android settings) to handle only calls, reception works well, but in transmission, it sends the audio to the smartphone instead of to the Bluetooth. FT8CN always shows the message "enter Bluetooth headset mode". How can I solve this? Thanks in advance for the suggestions
[image]

temskiy, 2025-06-10:
Try to use SoundAbout app. Configuration:
Media Audio -> Bluetooth (mono)
Phone calls -> Bluetooth (mono)

Generally it works, but sometime with freezes :(

lordiu, 2025-12-16:
I have exactly the same behavor here. please fix

version 0.93
samsung sm-a505 + honor jsn-lx1