N0BOY/FT8CN #123: Problems on FT8CN 0.92 with (tr)uSDX

dk5cr, 2024-09-02:
Hi, 
i connected the (tr)uSDX to my new Redmi13C and my older Galaxy A5 (2017), installed FT8CN 0.92 on it, started the app and connected the USB-cable. 
Cellphone asks for serial connection and establishes the connection. (tr)uSDX switches to the preselected QRG and it "clicks" two times. So far so good, but after a fraction of a second the following error occurs on the cellphone: 

[image]

After this message, adjustment in the app is possible, but will not be transferred to the (tr)uSDX so it seems that serial connection is lost. After replugging both devices the same question (serial port) appears, clicking, error message... 

I tried with the two cellphones and a tablet with different android versions, none of them showed an other behaviour, just the same issue. 
HELP :) 

Thanks and 73 de Cornel,  DK5CR 

PS: Please don't paste answers in Chinese letters, i'm not able to read/understand. Thx

dc1rdb, 2024-09-02:
Hi Cornel,

Just my two cents:
Do you have the latest firmware on your (tr)uSDX? Should be R2.00x
Did you try a different USB cable?

73 de Robert

dk5cr, 2024-09-03:
Hi Robert,
thanks for the reply. Of course 2.00x is on the truSDX and I tried different cables. Band switch works for first second when starting FT8CN and it confirms its connection to truSDX. But then come the other messages immediately, even with other USB cables. 

Thanks and 73 de Cornel, DK5CR

dc1rdb, 2024-09-03:
Hi Cornel,
Very strange indeed. When I use Google Lens on your screenshot, the Chinese symbols translate to "with serial port lost communication". I have no clue what "2>1" means in this context. 
I am using (tr)uSDX with ft8cn app quite regularly with my tablet or my phone (both Android11) and I never had this error message.
Let's hope that N0BOY can enlighten us.

73 de Robert

dk5cr, 2024-09-03:
Hello Su and Robert, 
my truSDX has classic board inside. By double tapping on the button it switches through 80/40/20/15/10 (did i miss one?). 
The antenna is adjusted to an SWR of 1.1 on 20 and 15, USB cable has ferrit coils on the cable to reduce unwanted waves mirroring back into the rig. 

To eliminate the truSDX as the source of  error I tried with my Icom 7300 and Icom 7100 with the same result. Rig will be connected, CAT clicks the relais and then FT8CN disconnects with the same error message. 

I also used other Android devices with lower versions like 13 and 10 and a tablet with Android 11, same rigs on the different phones, but same error. 

It would be great to have a switch for logging errors in FT8CN with a high (or variable) verbose level to get more information about the reasons for losing serial connection... 

Cheers and 73 de Cornel, DK5CR

dc1rdb, 2024-09-03:
On my devices, FT8CN only asked for microphone and location permission, which I both granted.

73 de Robert

dk5cr, 2024-09-03:
Thanks for your reply, Robert. That's what my permissions are too.... So we could exclude this as reason for the error.

73 Cornel, DK5CR

d3m3vilurr, 2024-09-04:
@dk5cr can you take your setting screen? 

there is two possibility
1. invalid [serial setup](https://github.com/N0BOY/FT8CN/issues/42#issuecomment-1748055947)
2. or another unknown issue; this issue was similar as @rmoravcik 's [case ](https://github.com/N0BOY/FT8CN/issues/42#issuecomment-1742000834) (so @rmoravcik might have an experience to support this.)

according [his last comment](https://github.com/N0BOY/FT8CN/issues/42#issuecomment-1766833656), he was tested in practice mode, but imo, it only works in `CW` mode.
in my very old memory,  when a rig boot up in `CW` mode, rig cannot connect with CAT mode properly.

so can you follow this step?

- boot up rig(connect power source/usb line or battery) without ft8cn
- change mode to `USB` mode if rig is in the any other mode
- change bandwidth to `4k0`
- change band & freq to 14.074(optional)
- then connect rig to phone
- open ft8cn

73 de DS1UFX

dk5cr, 2024-09-09:
Hi, 
sorry for late response, was a lot of trouble here at home :-( 

To your concern: 
1) Where is the serial setup and where i can set preferences to the serial interface in FT8CN or truSDX? 
The Chinese signs are not readable for me, so i've got no clue where to look at serial settings "usbSerialPort.setParameters"... 

2) Moravics topic is not mine due to Firmware 2.00x, so in the newer firmware these issues were obviously fixed by DL2MAN as far as i've understood. 

So in the meantime i checked all settings in the truSDX and one setting was unknown for me, it named something with "QSK". It was enabled or active or so and i deactivated the entry. 
Then i purchased a direct cable from USB-C to USB-micro without any adapters or so. 
As a third try i installed some ferrits on the cable and suddenly **it worked.** 

Sometimes, when cellphone or USB-cable is too close to the RG-58 wire, the same error occurs again. And on lower (80/40/30) bands it's more often than in 20m. 

So last Sunday i was able to have 11 QSO with nearly 5W with the lowbands-board in 20m and 40m (just broiled my MosFet FDT86286 on the Classic-bands-Board! Any idea where to purchase a bunch of spare MosFets?) 

I would appreciate any hints to a source, where to purchase the FDT86256.
But the issue itself is solved. Thanks for helping 

Cheers and 73 de Cornel, DK5CR

pengowray, 2024-09-09:
Sorry to repeat another response to which you've already replied. I just want to say after I updated my (tr)uSDX to the "latest" firmware, I had the exact same error message, and the same issue (briefly seeming to work and then disconnected). However, there's the latest firmware and there's the [latest experimental Beta release](https://dl2man.de/wp-content/uploads/2022/01/wp.php/beta.html) which are found at different links. The latest experimental Beta release is what was needed.

d3m3vilurr, 2024-09-10:
hi.

>     1. Where is the serial setup and where i can set preferences to the serial interface in FT8CN or truSDX?
>        The Chinese signs are not readable for me, so i've got no clue where to look at serial settings "usbSerialPort.setParameters"...

nvm. I just want to see your setting options of the FT8CN, but i don't need it that because you already resolved your problem. :)

> So in the meantime i checked all settings in the truSDX and one setting was unknown for me, it named something with "QSK". It was enabled or active or so and i deactivated the entry. Then i [...]

Pretty much it's RFI issue between rig and antenna.
when i use EFHW 20m wire with 49:1 transformer, sometimes i can't operate 15~10m band on Xiegu x6100.
it's highly related setup environment, such as antenna position, rig position, feed wire length, using USB isolator, ferrite choke, CMC, grounding, counterpoise, ...

anyway congrats made QSO :+1:

> So last Sunday i was able to have 11 QSO with nearly 5W with the lowbands-board in 20m and 40m (just broiled my MosFet FDT86286 on the Classic-bands-Board! Any idea where to purchase a bunch of spare MosFets?)

when I was burned my MOSFET, I just replaced 3 BS170. BTW if you burned FDT86286, you can find it on [mouser][1] or [digikey][2]

[1]: https://www.mouser.de/ProductDetail/onsemi-Fairchild/FDT86256?qs=EN3VjTIJKYEcSkHvgYzvuA%3D%3D
[2]: https://www.digikey.de/de/products/detail/onsemi/FDT86256/3025424?s=N4IgTCBcDaIGYBMAuAOAbGArGkBdAvkA

73 de SU, DS1UFX

dk5cr, 2024-10-01:
Hello folks, 
few days my new Redmi13c worked properly with my truSDX with firmware 2.00x. I worked a lot of stations in Europe and overseas, even with only 5W. 
Last session worked well on Sunday. Yesterday i tried the same setup with the same cables, the same cellphone, the same truSDX, same same accumulatores, but... 

truSDX connects with my cellphone, FT8CN asks for serial port 
\0c03F5\0x1A86\A86\0x7523\0x0
(seems like this nummer changes from time to time... ) 
truSDX clicks two times and gets silent, but after few seconds truSDX starts to make noise again and FT8CN shows no audio spectrum anymore. For a fraction of a secont, spectrum is visible, but then truSDX clicks again and it's gone.  

Settings seems to be all right, USB CAT connect via truSDX (audio via cat), 1152ßß Baud, CI-V A4 (tried also 0, but without success). 
Just flashed the firmware 2.00x on the truSDX and tried again, but without any change. 

I really don't believe, rigs have an own soul or so, but in this case, it seems to be like that ;-) 

Do you have any proposals? 
Thanks and 73
Cornel, DK5CR

dk5cr, 2024-10-10:
hi, no one with an idea to solve this issue? I didn't update or upgrade anything, configuration was exactly the same as few days before, but after connecting truSDX to the Redmi13c, the orange box (trusdx) clicks two times and afterwards audio comes back thru the trusdxs speaker :-( 
CAT commands like send or QSY are working properly... HELP ;-)

kolbma, 2025-02-02:
Here, I've to close the app. And restart it, then the serial connection works again.  
Could be that it never works at the 1st time, but 2nd.  
But overall there is some instability. It runs sometimes very long. And from time to time there is connection problem and you have to reconnect.  
Not sure if the problem is in the communication of the app or some firmware bug of truSDX 2.00x.

I've had it already working once, and only once, with enabled speaker. Would be nice to have the option to listen to the audio.  
So seems to have been some kind of sync problem.