N0BOY/FT8CN #138: Unwanted Target Switchuing

ZB2ham, 2025-02-26:
I have had this problem since I started using FT8CN almost two years ago. It got better, but now it's getting so severe that most times I just give up in desperation and switch off.

**FT8CN DECIDES WHAT SIGNAL IT RESPONDS TO, OVERRIDING MY SELECTION OF MESSAGE AND TARGET**

I have copied and pasted the text below from https://www.reddit.com/r/amateurradio/comments/1f14xop/ft8cn_issue_unwanted_target_switching/?rdt=57858   
Quite obviously I am not alone in having this problem;

###########################
I don't know the correct nomenclature for this, so I apologize for the title. I was on a POTA/SOTA attempt today, trying to use FT8CN for the first time. I checked that it could CAT control my ic7300 (believe me, I know... I carried it), and decided that everything else could be sorted on the fly. I was wrong.

Everything seems to work well, except that the following would happen (broken down by 15 sec cycle):

I'd call CQ pota

Station A, B, C, D, and E would send a signal report.

FT8CN would select and respond to E

Station E would send confirmation, and A and C would resend their signal report

FT8CN switches to C, since they are the last caller in the log, but I haven't finished my QSO!

I struggled with this the whole time. I couldn't even call CQ most of the time, because the app was trying to respond to something earlier in the log. Please help! I tried a bunch of settings to no avail.
##############################

When you have a slightly rare prefix, pile-ups are common and FT8CN always chooses to reply to the last callsign in the list of stations calling you, you cannot override, you have NO CONTROL AT ALL. It spoils what should be a pleasant activity.

**PLEASE PLEASE PLEASE can we have this auto-QSO function disabled, perhaps with a toggle switch so that those who want it can have it, and those who do not want it can go fully manual?**

Many thanks

Ken ZB2MD

devinberg, 2025-03-14:
Changing "Auto call tracked" to "No call tracked" in the settings should fix this. Or at least it did for me.

ZB2ham, 2025-03-17:
Many thanks for replying.

I have had the "No Call Tracked" option selected for a long time, and it still happens. 

This morning I had a small pile-up and I had to switch off, because I couldn't work the station I wanted to for this problem. When I have several stations calling, FT8CN ALWAYS chooses to reply to the last station in the list, regardless of the state of the QSO I'm in.

The only way to avoid it replying to any particular station is to include it in the "Excluded callsign prefixes" list, but than at the cost of contacting ANY station from whatever country that was from. Hardly a solution

73s, Ken