N0BOY/FT8CN #140: QRZ API won't work

dk5cr, 2025-04-10:
Hi folks, 
i've got a QRZ API and put the Code into FT8CN. After clicking on Test, the app always tells me "Fail!", even if the API Code is correct. 

What's my fault? I type it in with - and it has chars and numbers in uppercase. Even without the - it won't work. 

Thanks for hints and clues. 

Cheers 
Cornel

zolee1988, 2025-04-15:
Hello,
Do you have xml subscription on qrz.com?

"Interacting with the QRZ Logbook through our API is a subscriber-only feature. The owner of the logbook must have an active subscription at the XML level or higher."

[https://www.qrz.com/docs/logbook30/api](url)

chrisdebian, 2025-07-22:
Do we know if this is still an issue? I'm looking to install FT8CN, in the next week or so.